TG9541/stm8ef #398: W1209 module with 5V

scientistnobee, 2021-02-01:
Hi, I would like to make a small heat pad with 5V battery powerbank using W1209. However, all I found is 12 V modules. The 5V modules advertised in ebay/Aliexpress don't seem to have genuine parts. Could you please advise me how to get the genuine 5V module or how to convert a 12V module into a 5V module. I have seen a similar question in electronics stack exchange, but there is no satisfactory answer is given, particularly about MOSFET and relay.

https://electronics.stackexchange.com/questions/544292/can-w1209-be-run-with-5v-dc-only

TG9541, 2021-02-01:
I had a look at the infos on stackexchange: 

* when the USB powerpack supply voltage is in the 4.5 to 5.5V range connecting the anode of D2 to the output of the 5V regulator should work. 

* After removing the 12V relay soldering a through-hole STP36NF06L at the same spot should work. It's maybe necessary to remove R7 so that the STM8S003 pin10 (PA3) doesn't get any additional load.

scientistnobee, 2021-02-01:
Thanks for your prompt reply. Is it okay to short input and output of the 12V regulator? Is there any link where I can buy genuine boards?

TG9541, 2021-02-01:
I think that shorting in and out should be OK. The thing I proposed would result in a slightly higher supply voltage.
Genuine boards have become as rare as hen's teeth. I asked several vendors but the supply side appears to be dead. The last time I got a board with an STM8S003 controller was nearly two years ago. Today it's almost easier to design your own board with KiCad and have JLCPCB assemble it for you.

scientistnobee, 2021-02-02:
I did some experiments with the module by just decreasing the supply voltage. I think circuit is working as it is without any shorting of voltage regulator, except the relay. What I am thinking is can I turn on 12v relay with 5v supply voltage, just by changing 2k resistor at the base of the transistor? Any suggestions?

TG9541, 2021-02-02:
@scientistnobee I'll first convert this issue into a [discussion thread](https://github.com/TG9541/stm8ef/discussions)

closed 2021-02-02 by TG9541