TG9541/stm8ef #400: Bug in UM/MOD?

MrMarkB, 2021-02-02:
I've found an apparent bug in the UM/MOD word.

Test case producing error is "HEX 2400 F4 A2C3 UM/MOD .S" which gives result "22C2 2" where "A243 17F" is expected.

For context I'm developing a word that configures the Timer/Counter 2 to produce a 50% duty cycle PWM at a specified frequency 1-64k Hz.  The error occurs in the calculation of the TIM2_ARR value as "16 MHz / request_frequency".  It appears to be manifest for request frequencies greater than or equal to 41666 Hz (A2C3) and not for lower request frequencies.

MrMarkB, 2021-02-02:
Neglected to add, error was seen on version 2.2.26 STM8S103F3 pre-built load running on a MinDev board.

TG9541, 2021-02-02:
Context: definition of UM/MOD:

```Forth
;       UM/MOD  ( udl udh un -- ur uq )
;       Unsigned divide of a double by a
;       single. Return mod and quotient.
```

the value of `un` is $A2C3 which, if `n` were the parameter, would be a negative number (b15=1). Using half this value (b15=0) gives a correct result: 

```Forth
HEX 2400 F4 5161 UM/MOD .S
 100 300 <sp  ok
```
I'm not sure if this is a limitation of the original eForth or a bug that I introduced. I will look into it.

TG9541, 2021-02-02:
I checked it: that's a limitation (alias "bug") in the original STM8EF code.  

Here is the minimal test that shows when the divisor gets too large:

```Forth
HEX
0 2 7FFF UM/MOD . drop 4 ok
0 2 8000 UM/MOD . drop 4 ok
0 2 8001 UM/MOD . drop 0 ok
```

I'll check if I can fix it.

MrMarkB, 2021-02-04:
I've attached a patched version of the UM/MOD code that I believe gets the expected results, albeit with limited testing.

The root issue is that the original algorithm does a comparison of the divisor (YTEMP) and the upper 16 bits of the shifted dividend (X) without considering the "17th bit" that was shifted out of X.  I've added a check for carry out and rearranged the logic a bit with the view of minimizing compiled executable size.

This seems to be working on a MINDEV board and in sstm8 simulator with a limited number of test cases.

[umMod.txt](https://github.com/TG9541/stm8ef/files/5926192/umMod.txt)

TG9541, 2021-02-04:
@MrMarkB thanks, that's of great value! I had started to analyze what's going wrong but couldn't find the time to apply a simulator, debugger or instrumentation. I'll include it in the source and run some tests cases. 

Would you like to fork, patch make a pull request? That way you can leave a "mark" in the code :-)

Picatout, 2021-02-06:
replaced UM/MOD in Picatout/stm8_eForth  and tested "HEX 2400 F4 A2C3 UM/MOD .S"
the result is: "2243 17F"  The most significant bit of remainder is missing.

Picatout, 2021-02-07:
bug corrected.   In you solution you used **SRAW X** at _MSMMb:_   but **RRCW X**  must be used here to gain most significant bit of remainder.

TG9541, 2021-02-07:
@Picatout merci mille fois!

I noticed that you used `DIVW X,Y` for the 16bit/16bit case. I've long had ideas about using `DIVW X,Y` for a 32bit/16bit UM/MOD using [this approach](https://stackoverflow.com/questions/55429707/32-bit-16-bit-signed-integer-division-without-32-bit-registers). This has never been a priority. Would it in your opinion be worth it?

VK6TT, 2021-02-07:
In my opinion any 16, 32 or 64 bit unsigned math routines are gold. The 
few routines I wrote in high level Forth to calculate values to load in 
a DDS (Direct Digital Synthesizer) look so cumbersome it's embarrassing.

Back when I lived in F-PC Mr Robert Smith wrote such a piece of code in 
assembler for the 8086 family. That and the references might help 
someone but I could never figure it out.

Kind regards

Richard

\ Division of unsigned quad by double.          08:45 31Oct87RLS

\ Version for sequential files and forward assembler.  Use with Tom Zimmer's
\   FF Forth.

comment:
 This file contains a Code (8086) routine for a true double
 precision divide routine  UMD/MOD .  The numerator is an
 unsigned quad precision number and the divisor is an unsigned
 double precision number.  The result returned is an unsigned
 double quotient (1st and 2nd on stack), and an unsigned double
 remainder (3rd and 4th on stack).  Another routine multiplies
 two unsigned double numbers, yielding an unsigned quad product.
 The file also contains a simple method for using labels in
 code routines in F83.

                  Robert L. Smith
                2300 St. Francis Dr.
                Palo Alto, CA 94303
comment;

comment:

      The divide routine is based on the article "Unsigned
      Division Code Routines" by Robert L. Smith (Forth
      Dimensions, Vol. VIII, No. 6, March/April, 1987), and
      on a subsequent letter "Mods Quad Divides" by Michael
      Barr (Forth Dimensions, Vol. IX, No. 2, July/August,
      1987).

      Note that the F83 assembler uses the sequence
             BX AX MOV
      to mean (in INTEL or MASM mnemonics):
             MOV AX,BX

comment;

comment:

\ Labels for Assembler                          03:54 02Nov87RLS

 30 CONSTANT MAXLABELS       HEX
CREATE SHORT_LABELS    MAXLABELS 4 * ALLOT
: SXBYTE   ( -- )  DUP 80 AND IF  FF00 OR  THEN ;   DECIMAL

: CLEAR_LABELS   ( -- )  SHORT_LABELS MAXLABELS 4 * 0 FILL ;

: CHECKLABEL   ( n -- m ) \ Or abort
     DUP MAXLABELS 1- U> ABORT"  Bad Label "
     2* 2* SHORT_LABELS + ;

: $     ( n1 -- n2 )
     CHECKLABEL DUP @
     IF  @  ELSE  2+ DUP @ SWAP HERE 2+ SWAP !
         DUP 0=  IF  HERE 2+ +  THEN
     THEN ;

comment;
[... 152 lines cut ...]
        MUL  DI                 \ AC
        ADD  AX, BX             \ AClo + ADhi + BC hi
        ADC  DX, BP             \ AChi + carrys
        MOV  BX, SI
        POP  SI
        POP  BP
        PUSH BX
        PUSH AX
        PUSH DX
        NEXT
        END-CODE

: D*    ( d1 d2 -- dprod )
     UMD* 2DROP ;

comment:

UMD/MOD    Unsigned division of a quad number by a double,
           yielding an unsigned quotient and remainder.
           If you wish to use Interrupt 0 for reporting
           errors, remove the parentheses from line 6.

comment;



Picatout, 2021-02-07:
@TG9541 ,
When I first adapted Mr thing code for NUCLEO-STM8S208  I changed the code  to use **divw x,y** in the case where **udh** is **0**.  But for the other cases I left it has the original.  After analyzing the problem I came to conclusion that there was no advantage of using **divw x,y**  in case where **udh** is not **0**.  This would require to much transfer between memory and registers. After the first division of **udh/un** it would be required to count the leading zero in **Y** (the remainder) subtract that number from 17 then save the partial quotient, put the remainder in X  put **udl** in **Y** shift left X:Y  (17-lzb) 
save **Y** again put **un** in **Y** again to do the next **DIVW X,Y**. The second part of quotient must be right appended to the first part.  And if there is bits left in **udl**  the process must be repeated.  This require to much step to be advantageous.

MrMarkB, 2021-02-08:
I'm a novice with github, so I apologize if I've made a mess of it, but I've tried to submit a pull request with the patches to UM/MOD as per posts by Picatout and myself above.  This should resolve issues 400 and 401 with the UM/MOD word.

I've put together a Python script to exercise UM/MOD from the serial port with random numbers and check the result.  With the patched build it hasn't called out any errors thus far.

closed 2021-02-08 by MrMarkB

TG9541, 2021-02-09:
@MrMarkB, you did pretty well for a first time pull-request!

I'm very happy about the contribution from both you and @Picatout! In hindsight more than one user (including I) had evidence that something is not quite right with UM/MOD. Maybe it's because signed operations rescued some of the defects through stripping bit15 of the divisor, and therefore no one ever looked closely enough. Furthermore, edge cases were never properly identitfied or tested (e.g. #401). Quite obviously, the art of testing numerical core routines needs more practitioners, so thanks a lot for your contributions!

@MrMarkB would you mind publishing you test code, maybe in a GitHub Gist?

TG9541, 2021-02-13:
@MrMarkB, @Picatout: this issue led to an interesting [discussion in the Forth2020](https://www.facebook.com/groups/2225595150855239/permalink/3703103453104394/?comment_id=3703227129758693&reply_comment_id=3711011715646901&notif_id=1612894596607545&notif_t=group_comment&ref=notif) FaceBook group: Bill Ragsdale remembered releasing a similar bug into the wild many years ago (which may well be the grandfather of these bugs here), and other stories about hunting down quite similar bugs emerged.

MrMarkB, 2021-02-15:
> @MrMarkB would you mind publishing you test code, maybe in a GitHub Gist?

@TG9541 If I've done this correctly, this is the Python3 code I used to exercise the um/mod word after making the patch.

The check is in the function "checkUmMod".  It takes three 16 bit integers as input, formats them into string for the STM8 eForth console to push them on the stack and execute um/mod and pop the quotient and remainder off the stack.  The console output is parsed to get the quotient and remainder the values of which are checked against Python's calculation of the same operation.

The main code simply generates random numbers in the range 0-FFFF for the parameters, calls the checkUmMod function, and prints the result if the result is not as expected.

<script src="https://gist.github.com/MrMarkB/da74e3d4f475ce60ccae4cff7c3bd2e9.js"></script>

TG9541, 2021-02-15:
@MrMarkB thanks a lot!

The Gist URL somehow isn't valid - don't know why, since this one, a fresh copy, doesn't look any different:
 
https://gist.github.com/MrMarkB/da74e3d4f475ce60ccae4cff7c3bd2e9

TG9541, 2021-02-21:
@Picatout you might find @JeeeK 's contribution interesting

commit 623305c0 (2021-02-09): Bugfix release fixes #400 UM/MOD
commit c3d755f3 (2021-02-09): Merge pull request #403 from TG9541/ummod

Bugfix release fixes #400 UM/MOD
commit 2f4bca47 (2021-03-09): fixes #400 minor UM/MOD improvement