TG9541/stm8ef #401: another bug in UM/MOD

Picatout, 2021-02-08:
I discovered another bug in UM/MOD and solved it. The **udh** part of the dividend must be smaller than the divisor to get the correct quotient but the original version was accepting **udh==un** to solve this bug I replaced **JRULE** for the overflow test by **JRULT**.  the modified code is:
```
UMMOD:
        PUSHW   X               ; save stack pointer
        LDW     X,(X)           ; un
        LDW     YTEMP,X         ; save un
        LDW     Y,(1,SP)        ; X stack pointer
        LDW     Y,(4,Y)         ; Y=udl
        LDW     X,(1,SP)        ; X
        LDW     X,(2,X)         ; X=udh
        CPW     X,YTEMP
        JRULT   MMSM1           ; udh must be < un to avoid overflow
; division overflow  return q=0,r=-1 
        POPW    X
        INCW    X               ; pop off 1 level
        INCW    X               ; ADDW   X,#2
        LDW     Y,#0xFFFF
        LDW     (X),Y
        CLRW    Y
        LDW     (2,X),Y
        RET
MMSM1:
        LD      A,#17           ; loop count
MMSM3:
        CPW     X,YTEMP         ; compare udh to un
        JRULT   MMSM4           ; can't subtract
MMSMa:
        SUBW    X,YTEMP         ; can subtract
        RCF 
MMSM4:
        CCF                     ; quotient bit
        RLCW    Y               ; rotate into quotient
        RLCW    X               ; rotate into remainder
        DEC     A               ; repeat
        JREQ    MMSMb           ; if A == 0
        JRC     MMSMa           ; if carry out of rotate
        JRA     MMSM3           ;
MMSMb:
        RRCW    X
        LDW     YTEMP,X         ; done, save remainder
        POPW    X
        INCW    X               ; drop
        INCW    X               ; ADDW   X,#2
        LDW     (X),Y
        LDW     Y,YTEMP         ; save quotient
        LDW     (2,X),Y
        RET
```

TG9541, 2021-02-08:
@Picatout that's something! Not two but three bugs in one core routine - something that almost always works is much worse than a plain bug.

@MrMarkB merged your new solution into the two points identified in #400

Merci beaucoup and thanks a lot to both of you!

TG9541, 2021-02-08:
@Picatout, @MrMarkB 

I just did a brief test - it appears to me that something isn't quite right when divisor bit15 is set: 

```Forth
0 2 $8000 um/mod . . 4 0 ok
0 2 $FFFF um/mod . . 0 0 ok
0 2 $7f00 um/mod . . 4 1024 ok
0 2 $8f00 um/mod . . 0 0 ok
```

TG9541, 2021-02-09:
Fixed - closing this. Please refer to follow-up discussion in #400.

closed 2021-02-09 by TG9541