TG9541/stm8ef #424: Interrupt and background data stacks coincide

Eelkhoorn, 2021-04-08:
The file inc/defconf.inc has two defines for the background data stack size:
        BG_STACKSIZE     = 32   ; Default size in bytes of data stack for background tasks
        BSPPSIZE         = 0   ; BG task data stack size (bytes) for memory calculation

BG_STACKSIZE is not used effectively in forth.asm.  Both the interrupt datastack and the background data stack grow down from $350 ( for a STM8Sx03 ).

Maybe BG_STACKSIZE is redundant and can be removed while BSPPSIZE is set to 32.
Or did I mis something here?

TG9541, 2021-04-08:
Confirmed - thanks a bunch for reporting this!

Applications that use the BG task *and* an ISR that uses Forth code may suffer from BG task data stack corruption. This quite obviously bad. I'll quickly provide a fix/

TG9541, 2021-04-08:
This happened when I factored out the BG code in #377. It was still OK until the 2.2.26 release. The first release that contained this bug was 2.2.27.pre1. 2.2.27 was released 9/Mar/2021 (today that's a month).

TG9541, 2021-04-08:
That's a Travis-CI / Docker Inc problem:
[image]

commit 349a14c0 (2021-04-08): fixes #424 BG task wrongly uses ISR data stack
closed 2021-04-08 by TG9541