TG9541/stm8ef #431: TIB behaviour

Eelkhoorn, 2021-04-21:
When TIB is linked in dictionary (by adapting  inc/defconf.inc: UNLINK_TIB       = 0 ) TIB returns the address of the terminal input buffer, as expected.
When TIB is linked afterwords with "#include TIB" TIB returns a pointer to the address of TIB. This is confusing.
I think this has changed somewhere between release 23 and 28 and breaks code written for earlier releases using TIB.

TG9541, 2021-04-21:
I guess that this is the same as #420 - could you please check if the release [2.2.28](https://github.com/TG9541/stm8ef/releases/tag/2.2.28) solves this issue?

Eelkhoorn, 2021-04-21:
I am afraid it does not. This is out/STM8L051F3/target/TIB from STM8eForth 2.2.28:
$8C46 CONSTANT TIB \ NVM STM8L051F3: addr. const. Terminal Input Buffer

Eelkhoorn, 2021-04-22:
CORRECTION. I checked with release 28 from april 5th. I just did a fresh clone and now it is solved:
$ cat out/MINDEV/target/TIB
$ $0350 CONSTANT TIB \ STM8L051F3: "TIB"  Term. Input Buf. TIBLENGTH between SPPLOC and RPP
My apologies. This can be closed.

closed 2021-04-22 by Eelkhoorn

TG9541, 2021-04-22:
The good thing is that this bug has been identified not only once but twice. Open source quality control works! Apologies for the bug and thanks for reporting it!