TG9541/stm8ef #439: Unable to configure external interrupt sensitivity.

Danya0x07, 2021-08-05:
Hello, thank you for your eForth system! I used it recently in my [project](https://github.com/Danya0x07/UVLT01).

I found out that it is not possible to write directly to EXTI_CR registers as to others. This is because it is only possible to write to these registers when global interrupts are turned off, but your system turns them on by default.

So, my workaround is that:

    [ $9B c, ]    \ ASM("SIM") - disable interrupts
    $30 EXTI_CR1 c!  \ PortC interrupt on
    [ $9A c, ] ;  \ ASM("RIM") - enable interrupts

May be you can describe this situation in project wiki or add some words that will turn global interrupts off for a piece of code?

Thanks:-)

TG9541, 2021-08-06:
Hi @Danya0x07

I was unaware that writing to `EXTI_CR1` requires interrupts to be off!

I'll add `SIM` and `RIM` words to the lib. 

Here is how it works:

```Forth
#i f/rimsim
Uploading: ./f/rimsim

\res MCU: STM8S103
\res export EXTI_CR1

  $50A0 CONSTANT EXTI_CR1 ok

: SIM $9B C, ; IMMEDIATE  ok
: RIM $9B C, ; IMMEDIATE  ok

: r OK
EXTI_CR1 C@ . ; ok

: PcIntOn OK
$30 EXTI_CR1 c! ; ok

: init OK
SIM PcIntOn OK
RIM ; ok
Closing: ./f/rimsim  ok 
r 0 ok
PcIntOn ok
r 0 ok
init ok
r 48 ok
```

By the way, cool project!

TG9541, 2021-08-07:
I added a section [Using the Pin-Change interrupt](https://github.com/TG9541/stm8ef/wiki/STM8-eForth-Interrupts#using-the-pin-change-interrupt) to the wiki and a reference to your project to the [example code](https://github.com/TG9541/stm8ef/wiki/STM8-eForth-Example-Code).

Thanks for reporting this :-)

closed 2021-08-07 by TG9541