TG9541/stm8ef #456: Clean up ADC code

TG9541, 2023-01-28:
STM8S / STM8L conditional code in `stm8_adc.inc` is a bit hard to read and Forth headers are duplicated. Clean it up.

TG9541, 2023-01-28:
The instruction `BRES    ADC1_CR1,#0     ; disable ADC` in `stm8_adc.inc` is unreachable.

```Asm
 
 ;       ADC@  ( -- w )
 ;       start ADC conversion, read result
 
         HEADER  ADCAT "ADC@"
 ADCAT:
         .ifeq   (FAMILY - STM8L)
 ;       ADC for the STM8L family
         BRES    ADC1_SR,#0      ; reset EOC
         BSET    ADC1_CR1,#0     ; enable ADC
         BSET    ADC1_CR1,#1     ; start ADC
 1$:     BTJF    ADC1_SR,#0,1$   ; wait until EOC
         LDW     Y,ADC1_DRH      ; read ADC
         JP      YSTOR
         BRES    ADC1_CR1,#0     ; disable ADC
```

@Eelkhoorn the STM8L `ADC@` code was provided by you. Should I fix and test something (maybe the intention was to conserve power)? Otherwise I'll just remove that line.

Eelkhoorn, 2023-01-29:
Hoi Thomas

I don't know what "The instruction ... is unreachable" means, but I
think that " BRES    ADC1_CR1,#0     ; disable ADC"  can be
removed harmlessly.

From AN3137, 2.3.4:
The ADC consumption (when the ADC is active and is converting) is given
in the device datasheet (around 1400 µA). In order to save power, it is
recommended to perform the ADC conversion at the highest possible
speed, and then switch the ADC off.

I think this is why I have put it there.

On Sat, 28 Jan 2023 12:06:17 -0800
Thomas ***@***.***> wrote:

> The instruction `BRES    ADC1_CR1,#0     ; disable ADC` in
> `stm8_adc.inc` is unreachable.
>
> ```Asm
>
>  ;       ADC@  ( -- w )
>  ;       start ADC conversion, read result
>
>          HEADER  ADCAT [...]

TG9541, 2023-01-29:
Hoi @Eelkhoorn, thanks, that explains it :-) 

Unreachable it is because the `JP      YSTOR` effectively ends the subroutine (it contains the `RET` instruction). 

The following would work:
```Asm
1$:     BTJF    ADC1_SR,#0,1$   ; wait until EOC
        LDW     Y,ADC1_DRH      ; read ADC
        CALL      YSTOR
        BRES    ADC1_CR1,#0     ; disable ADC
        RET
```

But then it's better to do this:
```Asm
1$:     BTJF    ADC1_SR,#0,1$   ; wait until EOC
        LDW     Y,ADC1_DRH      ; read ADC
        BRES    ADC1_CR1,#0     ; disable ADC
        JP      YSTOR
```

Since more tests are needed in order to optimize power consumption in an application, the line will be removed for now. Control of the peripheral and the clock setting can also be done in the application.

closed 2023-01-29 by TG9541