TG9541/stm8ef #457: Strange behaviour of LITERAL

Eelkhoorn, 2023-01-28:
When used within a "if then" block LITERAL behaves strange.

HEX ABCD DUP
: test1 LITERAL ;
: test2 IF LITERAL THEN ;
test1 .  ABCD ok        ( this is expected)
-1 test2 . B4 ok         ( ???)

TG9541, 2023-01-28:
Hoi @Eelkhoorn! 

I'm not sure if that's what you intend to do... but the following code will show a more consistent behavior:

```Forth
HEX ABCD
: test2 IF [ SWAP ] LITERAL THEN ; 
```

The result is now consistent with the input: `-1 test2 . ABCD ok`.

Without the `SWAP` you'd be using the address of the dummy `IF` jump target (patched by `THEN`).

Eelkhoorn, 2023-01-28:
Hoi Thomas

OK, that makes sense. Now I understand what happens here. I wanted to
make a vector table, like:

here ( start of table)
' some_word ,
' some_other_word ,
...
...
' last_word ,

: execute_vector ( n f--)
   IF [ SWAP ] 2* LITERAL + @ EXECUTE THEN
;

So, it is no issue at all, just my poor understanding of IF THEN
compilation. Thank you very much for the explanation.

I will look at the other issue (#456) tomorrow. I don't directly recall
why I came up with that. It might indeed be for power consumption
reasons, I made a small solar battery charger with a STM8L that uses
ADC and DAC some years ago.

 On Sat, 28 Jan 2023 10:30:34 -0800
Thomas ***@***.***> wrote:

> Hoi @Eelkhoorn!
>
> I'm not sure if that's what you intend to do... but the following
> code will show a more consistent behavior:
>
> ```Forth
> HEX ABCD
> : test2 IF [ SWAP ] LITERAL THEN ;
> ```
>
> The result [...]

closed 2023-01-28 by TG9541

TG9541, 2023-01-29:
@Eelkhoorn maybe the [FC-table](https://github.com/TG9541/stm8ef-modbus/blob/d07bc689a607f71386d58a0bd9205a99f41b83bb/MBPROTO#L36) implementation in [STM8-Modbus MBPROTO](https://github.com/TG9541/stm8ef-modbus/blob/master/MBPROTO) matches your use case:  it allows patching a table at runtime and it's well tested. [FC>XT](https://github.com/TG9541/stm8ef-modbus/blob/d07bc689a607f71386d58a0bd9205a99f41b83bb/MBPROTO#L41) is the access word and [@?EXEC](https://github.com/TG9541/stm8ef-modbus/blob/c627589f8e8982d0d2f1c5a760fd9d130c9aa8e0/MBBASE#L46) the execution word. You'll also find an extension mechanism with a default word.

Eelkhoorn, 2023-01-29:
Hoi Thomas

That is very interesting, thank you. 
I am currently trying to integrate stm8ef in muforth. I intend to leave
the interpreter and compiler out of the kernel to save flash.

On Sat, 28 Jan 2023 21:40:06 -0800
Thomas ***@***.***> wrote:

> @Eelkhoorn maybe the
> [FC-table](https://github.com/TG9541/stm8ef-modbus/blob/d07bc689a607f71386d58a0bd9205a99f41b83bb/MBPROTO#L36)
> implementation in [STM8-Modbus [...]

TG9541, 2023-01-29:
Wow, that sounds interesting! If you get it working please share!

Eelkhoorn, 2023-03-27:
Here is the first attempt of using muforth to take over the
interpreting and compiling functions of stm8ef. I tested it with two
targets: MINDEV and W1209-FD. It is probably full of bugs, comments
and suggestions for improvement are welcome. 
I tried it on a W1209-FD application for a simple timer, with the menu
structure copied from the logging thermostat project, and it works
fine. The flash space reduction is significant: kernel takes 2179
bytes, the application takes 1213 bytes, 4799 bytes left.

https://github.com/Eelkhoorn/stm8ef-mu

On Sun, 29 Jan 2023 12:17:09 -0800
Thomas ***@***.***> wrote:

> Wow, that sounds interesting! If you get it working please share!
>

TG9541, 2023-03-28:
I'm looking forward to testing https://github.com/Eelkhoorn/stm8ef-mu !