TG9541/stm8ef #464: ES7P003FGTF support

rtek1000, 2023-04-03:
Hello,

Do you know how to program the ES7P003?  Tools and IDE? I received a W1209 board with this microcontroller.

Maybe you can add the ES7P003FGTF on this page:

https://github.com/TG9541/stm8ef/wiki/W1209-Identifying-'compliant'-boards

ES7P003 (The board does not have Vcap)
[image]

STM8S003
[image]

Datasheet (Automatically translated):
https://github.com/rtek1000/W1209-firmware-modified/blob/master/Doc/ES7P003FGTF_EN.pdf

Datasheet original:
https://github.com/rtek1000/W1209-firmware-modified/blob/master/Doc/ES7P003FGTF_CN_622163.pdf

rtek1000, 2023-04-03:
Oh, I forgot to comment, the W3230 board also has an STM8S003F3P6.

Thanks.

rtek1000, 2023-04-03:
I found the manufacturer's website, it seems that everything is available for download, from the compiler to the programmer (ES-Link 😂) and via the serial port.

https://www.essemi.com/index/product/detail?id=39

rtek1000, 2023-04-03:
Sadly, QFN24 is also used on the W1209:

[image]

https://electronics.stackexchange.com/questions/591867/can-anyone-identify-the-microcontroller-on-a-w1209-thermostat

VK6TT, 2023-04-04:
Hi,

I didn't see anything to suggest the device is code compatible with the 
STM8 devices. Perhaps only having one good eye at present meant I missed 
it. However, this would make it hard for anyone on this mailing list to 
offer suggestions on how to program since we are focussed on the STM8 
series devices.

Best regards

Richard

TG9541, 2023-04-04:
Interesting, that's the 3rd µC in a STM8S103 TSSOP20 form factor I get to know about (the others being N76E003AT20 - MCS-51 and CH32V003F4P6 - RISC-V). 

This one appears to have a very basic proprietary 8 bit core with (fixed?) 16bit instruction width:

[image]

At least part of the instruction set is memory access controlled (e.g, multiplication, division):
[image]

I doubt that this chip is more than a stop-gap to serve the domestic market. I would rather not try mastering this one.

A complete datasheet is here:
https://datasheet.lcsc.com/szlcsc/1912111437_Qingdao-Eastsoft-Communication-Tech-ES7P003FGTF_C394846.pdf

rtek1000, 2023-04-04:
I didn't expect to be able to use the same firmware, but the logic can be reused.

I am interested in using the W1209 board with DS18B20 sensors, I currently use Attiny85 with [this configuration](https://github.com/rtek1000/ATtiny85_DS18B20_Thermostat), but I have to assemble the board manually.

Incredibly ES7P003 has support in VScode ([Extension Essemi](https://marketplace.visualstudio.com/items?itemName=luh.esext)), I'm trying if I can make it work, the documents are in Chinese. I am using translator to read and access the website. :sweat_smile:

But I'm a little wary of VScode, because Platformio generated a lot more code in VScode with SDCC than STVD with Cosmic, commented [here](https://github.com/platformio/platform-ststm8/issues/60).

TG9541, 2023-04-04:
I guess it would be worth thinking about producing an open-source STM8 board that can be ordered, e.g., from JLCPCB at a budget:

* a form factor that fits in a standard enclosure
* through-hole assembly options, e.g., for headers, relays and a LED display
* e.g., STM8SxxxKyT6C µC (e.g., STM8S103K3, STM8S903K3, STMS105K6...), OR maybe the TSSOP20 form factor so that optionally any of the other chips can be used (the CH32V003F4P6 RISC-V sure looks interesting).

rtek1000, 2023-04-04:
Yes, I've already ordered from JLCPCB, it's a very good option.

rtek1000, 2023-04-04:
What about the noise immunity of the STM8? Atmel's original ATmega chips were much better than Microchip's PICs.

VK6TT, 2023-04-04:
Don't spend any time on this Thomas because I doubt the time and effort 
is warranted. It was mid 2019 when I had a derivative of the W1209 board 
made and I have never found a need to pick one up again.

A library of gerbers might be the simplest way. If anyone wants the 
gerbers, pick and place file and BOM file for this board they only need 
ask.

The board was used on top of another board containing a number of large 
power fets and was used as a voltage sensitive relay.

I only mention this because it is a real leap to go from something that 
is essentially one-off for your own use to a PCB with wider appeal. When 
I think about all the STM8 projects I have built and the absence of 
requests for a PCB or details on how to have one built I'm convinced 
your time could be better spent elsewhere. OR have you had success with 
other people wanting to emulate on of your projects?